stb-tester/stb-tester #246: get_frame provides an old frame after a sleep while video wasn't rolling

szemate, 2015-01-06:
`stbt._display.last_sample` stores the most recent GSt buffer until `stbt.gst_samples` removes it or it's replaced by a newer buffer. To put it other way, if the video stream stops then `gst_samples`
yields the very last sample of the video stream even though it's called minutes later.

This behaviour has unwanted effects in the following situations:

1.
```py
stbt.press("POWER_OFF")
time.sleep(5)
# Make sure that the device under test is off
try:
    stbt.get_frame()
except stbt.NoVideo:
    pass
else:
    raise stbt.UITestFailure("The device is still on")
```
In this case `stbt.get_frame` never raises `NoVideo`, as it returns the
sample that had been stored *before* the box was powered off.

2.
```py
stbt.wait_for_match("menu.png")
stbt.press("POWER_OFF")
time.sleep(5)
stbt.press("POWER_ON")
# Make sure that we no longer see menu.png because
# it should disappear if the device under test is restarted
try:
    stbt.wait_for_match("menu.png")
except stbt.MatchTimeout:
    pass
else:
    raise stbt.UITestFailure("menu.png hasn't disappeared")
```
In this case we have a race condition between stbt and the device under
test. The device needs some time to start delivering frames after it's
powered on; if it doesn't happen before `gst_samples` is called then
`wait_for_match("menu.png")` returns false positive as the first sample
that `gst_samples` yields is the one that had been stored *before* the
device was powered off.

drothlis, 2017-07-24:
There's the more general issue of video-capture latency. For example in `wait_for_transition` (see #445) we do something like this:

```
stbt.press("KEY_DOWN")
press_time = time.time()

# Wait for animation to start
while True:
    f = stbt.get_frame()
    if f.time < press_time:
        continue  # Discard frame to work around latency in video-capture pipeline
    ...
```

I'm not sure what to do about this in the general case. Perhaps `stbt.get_frame` and `stbt.frames` could have similar logic as the above, and discard frames older than the time at which `get_frame` was called. But that would probably break things for people with badly-configured source pipelines (with incorrect timestamps). I don't know how common such bad configurations are, but it's not trivial to get right, so probably quite common.

wmanley, 2017-07-24:
>Perhaps `stbt.get_frame` could have similar logic as the above

I'm pretty sure we wouldn't want this.  If your video-capture latency was 100ms the maximum framerate you could get from calling `stbt.get_frame()` in a loop would be 10fps.

> Perhaps `stbt.frames` could have similar logic as the above

I think there's a stronger argument for this, but I'm still unsure.

An alternative would be to build this logic into `stbt.press` - the theory being that you wouldn't want frames from before the last time you interacted with the device.  We could potentially also add a `stbt.sleep` which does something similar.  This would make test-scripts easier to read because it would ensure an ordering of events.  e.g.

    stbt.press('KEY_OK')
    stbt.wait_for_match('something')
    stbt.sleep(5)
    stbt.wait_for_match('something_else')

You would then know that things are happening in that order, which might reduce surprises.

As an aside: I'm reminded of two things relating to asynchronous events:

* [Nagle's Algorithm](https://en.wikipedia.org/wiki/Nagle%27s_algorithm) - groups together `send`s on a socket, but sends immediately once the program switches to calling `recv`
* [Some thoughts on asynchronous API design in a post-async/await world](https://vorpus.org/blog/some-thoughts-on-asynchronous-api-design-in-a-post-asyncawait-world/) - discusses making async programming with `yield` more predictable by guaranteeing "causality".  "causality" is what we want here too I guess.

drothlis, 2017-07-24:
It's good to get these thoughts documented for future reference.

For @szemate's original issue, I think you should either fix your source pipeline so it continues delivering (black) frames while the device-under-test is powered off, or check the frame times like `wait_for_transition` does.

closed 2017-07-24 by drothlis