stb-tester/stb-tester #247: Unpredictable timeout if source pipeline restarts during wait_for_match

szemate, 2015-01-06:
Every time the source pipeline is rebuilt, timestamps of the video buffers restart counting from zero.

If the source pipeline restarts while `gst_samples` is running, the actual timeout value is getting unpredictable, as `gst_samples` times out if `timestamp - self.start_timestamp > timeout_secs`, and both `timestamp` and `self.start_timestamp` are reset to 0 while `timeout_secs` stays the same. (https://github.com/drothlis/stb-tester/blob/master/stbt/__init__.py#L1749)

E.g. if we are running `wait_for_match('something.png', timeout_secs=20)` and the source pipeline restarts after 10 seconds, it raises `MatchTimeout` after ~30 seconds.

wmanley, 2015-01-06:
We really shouldn't be using buffer timestamps directly in stbt.  Technically-speaking they're meaningless without a `GstSegment` to interpret them into running-time or stream-time.

Perhaps we can start dealing with stream-time and manipulate GStreamer to set stream-time to wall-clock time when we create the pipeline in the first-place.  That way stream time would start being more meaningful than "time since we started capturing video".  I don't know what effect that would have on the sink-pipeline.  It would probably be fine.

wmanley, 2015-01-06:
Hmm, maybe setting stream-time to wall-clock time is meaningless as we can send a SEGMENT event in PAUSED state and you'd expect the next buffer out to be timestamped with the time in the SEGMENT event?  Not sure.

closed 2016-06-27 by wmanley