stb-tester/stb-tester #248: Incorrect "Blackmagic card froze" classification

LewisHaley, 2015-01-15:
I've seen the "Blackmagic card froze" classification be incorrectly given with the captured decklinksrc.log looking like:
```
[2015-01-14 18:08:00.402810 +0000]  run: Checking Blackmagic video-capture device
[2015-01-14 18:08:02.109335 +0000]  0:00:01.593533503  5119      0x21b0070 INFO             decklinksrc gstdecklinksrc.cpp:364:gst_decklink_src_send_event:<decklinksrc0> EOS pending
[2015-01-14 18:08:02.109422 +0000]  0:00:01.593598653  5119      0x218e450 INFO             decklinksrc gstdecklinksrc.cpp:770:gst_decklink_src_task:<decklinksrc0> EOS pending
[2015-01-14 18:08:02.109448 +0000]  0:00:01.593617981  5119      0x218e450 INFO             decklinksrc gstdecklinksrc.cpp:932:gst_decklink_src_task:<decklinksrc0:videosrc> pushing EOS event
[2015-01-14 18:08:02.109470 +0000]  0:00:01.593638790  5119      0x218e450 INFO             decklinksrc gstdecklinksrc.cpp:934:gst_decklink_src_task:<decklinksrc0:audiosrc> pushing EOS event
```

When the subsequent test starts, the Blackmagic card works correctly. I *think* that the error is that the `sleep 1` is sometimes not long enough for the card to activate and thus "VideoInputFrameArrived" is not printed by the Gstreamer debug. I going to run some tests to see if I can reproduce this.

LewisHaley, 2015-01-15:
I was unable to reproduce this using a loop of launching the same code to capture gst debug as stbt-batch.d/run does in 400 iterations. In real usage the error comes up more often than that. It might be due to the fact that the check_hardware function only runs if NoVideo causes the test to fail, which changes the situation slightly.

drothlis, 2015-02-18:
@LewisHaley what do you think the solution is then -- increase the sleep?

LewisHaley, 2015-02-23:
I'm now seeing some more output in that same time but still with the wrong classification:
```
run: Checking Blackmagic video-capture device
0:00:00.023529163 21648      0x2c68960 DEBUG            decklinksrc gstdecklinksrc.cpp:610:gst_decklink_src_audio_src_query:<decklinksrc0:audiosrc> query: caps query: 0x2c5ca30, GstQueryCaps, filter=(GstCaps)NULL, caps=(GstCaps)NULL;
0:00:00.023570797 21648      0x2c68960 DEBUG            decklinksrc gstdecklinksrc.cpp:610:gst_decklink_src_audio_src_query:<decklinksrc0:audiosrc> query: caps query: 0x2c5cca0, GstQueryCaps, filter=(GstCaps)NULL, caps=(GstCaps)NULL;
0:00:00.023593315 21648      0x2c68960 DEBUG            decklinksrc gstdecklinksrc.cpp:635:gst_decklink_src_video_src_query:<decklinksrc0:videosrc> query: caps query: 0x2c5ccf0, GstQueryCaps, filter=(GstCaps)NULL, caps=(GstCaps)NULL;
0:00:00.023619804 21648      0x2c68960 DEBUG            decklinksrc gstdecklinksrc.cpp:635:gst_decklink_src_video_src_query:<decklinksrc0:videosrc> query: caps query: 0x2c5cb70, GstQueryCaps, filter=(GstCaps)NULL, caps=(GstCaps)NULL;
0:00:00.023711637 21648      0x2c68960 DEBUG            decklinksrc gstdecklinksrc.cpp:635:gst_decklink_src_video_src_query:<decklinksrc0:videosrc> query: caps query: 0x2c5cca0, GstQueryCaps, filter=(GstCaps)NULL, caps=(GstCaps)NULL;
0:00:00.024380358 21648      0x2c68960 DEBUG            decklinksrc gstdecklinksrc.cpp:635:gst_decklink_src_video_src_query:<decklinksrc0:videosrc> query: caps query: 0x2c5ca30, GstQueryCaps, filter=(GstCaps)NULL, caps=(GstCaps)NULL;
0:00:00.025037269 21648      0x2c68960 DEBUG            decklinksrc gstdecklinksrc.cpp:635:gst_decklink_src_video_src_query:<decklinksrc0:videosrc> query: caps query: 0x2c5ca80, GstQueryCaps, filter=(GstCaps)NULL, caps=(GstCaps)NULL;
0:00:00.025687938 21648      0x2c68960 DEBUG            decklinksrc gstdecklinksrc.cpp:635:gst_decklink_src_video_src_query:<decklinksrc0:videosrc> query: caps query: 0x2c5c940, GstQueryCaps, filter=(GstCaps)NULL, caps=(GstCaps)NULL;
0:00:00.026385695 21648      0x2c68960 DEBUG            decklinksrc gstdecklinksrc.cpp:635:gst_decklink_src_video_src_query:<decklinksrc0:videosrc> query: caps query: 0x2c5cb20, GstQueryCaps, filter=(GstCaps)NULL, caps=(GstCaps)NULL;
0:00:00.032659590 21648      0x2c68960 DEBUG            decklinksrc gstdecklinksrc.cpp:419:gst_decklink_src_start:<decklinksrc0> start
0:00:00.059952693 21648      0x2c89c50 DEBUG            decklinksrc gstdecklinksrc.cpp:746:gst_decklink_src_task:<decklinksrc0> task
0:00:01.595178568 21648      0x2c68960 DEBUG            decklinksrc gstdecklinksrc.cpp:359:gst_decklink_src_send_event:<decklinksrc0> handling event 0x2c86460 eos event: 0x2c86460, time 99:99:99.999999999, seq-num 99, (NULL)
0:00:01.595218698 21648      0x2c68960 INFO             decklinksrc gstdecklinksrc.cpp:364:gst_decklink_src_send_event:<decklinksrc0> EOS pending
0:00:01.595280064 21648      0x2c89c50 INFO             decklinksrc gstdecklinksrc.cpp:770:gst_decklink_src_task:<decklinksrc0> EOS pending
0:00:01.595298431 21648      0x2c89c50 DEBUG            decklinksrc gstdecklinksrc.cpp:914:gst_decklink_src_task:<decklinksrc0> pausing task, reason eos
0:00:01.595321652 21648      0x2c89c50 INFO             decklinksrc gstdecklinksrc.cpp:932:gst_decklink_src_task:<decklinksrc0:videosrc> pushing EOS event
0:00:01.595359421 21648      0x2c89c50 INFO             decklinksrc gstdecklinksrc.cpp:934:gst_decklink_src_task:<decklinksrc0:audiosrc> pushing EOS event
```

I think it is interesting that the `eos` is ~2s after the `start` yet still no frames are received.
However, I can't think of any other solution that increasing the sleep duration.

drothlis, 2017-07-24:
I think it would be a good idea to move classification outside of `stbt batch` anyway, for more flexibility and user-defined classifications. Unless someone provides a fix I'm tempted to remove the "Blackmagic froze" classification from stbt altogether (does anyone still use it?).

closed 2017-07-24 by drothlis