stb-tester/stb-tester #257: Test report page breaks if the contents of a cell are too long

szemate, 2015-02-17:
The javascript of `index.html` breaks with an error in `tablequeryjs` if any of the cells of the test results' table is longer than ~150 characters. As a result it is not possible to open test run details. This issue is easy to reproduce by inserting a long text into the "notes" column and reloading the page.

drothlis, 2015-02-17:
Thanks for the bug report.

* What was the error?
* What version did you test? (I want to know if this is a regression since v0.21.)

aiman, 2015-02-17:
Issue was reproduced on stbt 0.21 but also known to affect stbt 0.18 so not a regression.

drothlis, 2015-02-17:
> Issue was reproduced on stbt 0.21

That doesn't sound right. In 0.21 the notes column is truncated to ~30 characters, so length of the notes field wouldn't be able to trigger this.

I couldn't reproduce this on 0.21 but I've reproduced it on master.

> What was the error message?

To answer my own question: The error message is completely unhelpful:

    TypeError: this._d is undefined     tablequery.min.js:44

Unfortunately there doesn't seem to be an un-minified version of tablequeryjs available from the CDN.

drothlis, 2015-02-17:
After jumping through a whole bunch of hoops (which I may document later if the exercise turns out to be useful) to get an unminified version, the error is in moment.js and is no more enlightening:

```
(function (undefined) {
[...]
    extend(moment.fn = Moment.prototype, {
[...]
        isValid : function () {
[...]
                    this._isValid = !isNaN(this._d.getTime());
[...]
```

drothlis, 2015-02-18:
It looks like this is related to the bug where tablequery tries to figure out the column types by looking at the contents of the first row (after the table header). I could only reproduce this if the long table cell was in the first row.

szemate, 2015-02-18:
> I could only reproduce this if the long table cell was in the first row.

Yes it seems that it only affects the first row. Sorry that I haven't noticed it.

aiman, 2015-02-18:
>  In 0.21 the notes column is truncated to ~30 characters

It is `extra-columns` in particular that we are having a problem with.

closed 2015-02-18 by drothlis

szemate, 2015-02-18:
Thanks for the quick fix!

drothlis, 2015-02-18:
> > In 0.21 the notes column is truncated to ~30 characters
>
> It is extra-columns in particular that we are having a problem with.

You're right -- In 0.21 we truncated notes but not extra-columns. I didn't realise that. Oh well now we don't truncate anything in html, we use CSS for the truncation (which allows tablequery to work on the entire text).

commit 3b444bf7 (2015-02-18): stbt batch summary report: Hardcode column types for tablequeryjs

Fixes #257.

tablequeryjs (the library we use for browser-side filtering of testrun
results) looks at the table's first row (after the headings) to
determine what each column's data type is. For this it uses momentjs,
a third-party library bundled by tablequeryjs.

Sometimes it gets it wrong. For example it thinks that a particular git
sha is a timestamp, and so you can't search for textual matches on the
"git sha" column.

Another (more serious) bug is that if the content of a table-cell
(in the first row after the headings) i [... cut]
PR #258 (2015-02-18): stbt batch summary report: Hardcode column types for tablequeryjs
tablequeryjs (the library we use for browser-side filtering of testrun
results) looks at the table's first row (after the headings) to
determine what each column's data type is. For this it uses momentjs,
a third-party library bundled by tablequeryjs.

Sometimes it gets it wrong. For example it thinks that a particular git
sha is a timestamp, and so you can't search for textual matches on the
"git sha" column.

Another (more serious) bug is that if the content of a table-cell
(in the first row after the headings) is longer than  [... cut]