stb-tester/stb-tester #264: 0.21: Camera calibration is not working (endless colour analysis)

mxd-rglas, 2015-03-03:
Dear stb-tester team,
I did a fresh install of Ubuntu 14.04 and stb-tester 0.21 and stb-tester-camera.
When trying to run  the camera calibration stbt --with-experimental camera calibrate I get stuck when trying to analyse the colours.ts.
I cleared the .cache/stbt videos in order to re-create the reference videos. The colours.ts video is recognized, it looks like the script reads in some values and plots it to the colours graph, but it does not stop at 50 (as it did in former versions), but keeps on running endless.
[#####################################]      102 / 50

I uploaded a video for reference:
https://11968737038841875939.googlegroups.com/attach/e5e13e943065fd20/out-1.ogv
Looks like a bug to me 

in 0.21.

Best,
Robert

wmanley, 2015-03-03:
This seems to have been caused by a4ecd43.  It's caught by the automated tests so I'm not sure how it slipped through the net into a release.  I guess the issue is that Travis only runs the tests that we can still run on Ubuntu 12.04 as that's all it supports, so it can't run the `stbt camera` tests.

I'll prepare a fix.

mxd-rglas, 2015-03-03:
Tried the fix with our setup and it works. Thanks a lot for this quick fix!

PR #265 (2015-03-03): Camera: fix endless analysis
Fixes #264.
closed 2015-03-03 by wmanley