stb-tester/stb-tester #266: Multi-threading issue with Python Threading Lib (0.21)

mxd-rglas, 2015-03-03:
Whenever I use mutli-threading in my stbt scripts (for detecting random errors that cannot be expected by the script, like a popup error after disconnecting the network cable manually), from time to time there seems to be race conditions about the video frame, if I got the error message right. The error is not reproducible 100% and happens only intermittent. Most probably this could be fixed by Lock, but I would assume the stbt internals themselves need to respect the Lock as well.

Is there a plan to support multi-threading and fix this?

*Exception*:
UnboundLocalError: local variable 'res' referenced before assignment
Traceback (most recent call last):
 File "/usr/lib/stbt/stbt-run", line 49, in <module>
    execfile(args.script)
 File "/usr/lib/stbt/stbt.py", line 613, in wait_for_match
    raise MatchTimeout(res.frame, _template_name(image), timeout_secs)  # pylint: disable=W0631,C0301

*Code example*:
import threading
self._stop = threading.Event()
self._thread = threading.Thread(name='threadname', 
		target=do_something_in_parallel, 
		args=(self._stop,1))
self._thread.daemon = True
self._thread.start()
# wait for something special to happen
self._stop.set() # Fire the stop event to stop the parallel action

def do_something_in_parallel(stop):
  while not stop.isSet():
    # do your parallel stuff

wmanley, 2015-03-03:
This sounds like a reasonable approach, but I'm not surprised that it doesn't work 100% as this is not a use case that we have any automated tests for in stb-tester itself.

drothlis, 2017-11-10:
This should be fixed by #449.

closed 2017-11-10 by drothlis