stb-tester/stb-tester #297: Librification

wmanley, 2015-08-12:
It would be good if stbt could be used as a well behaved library, rather than requiring `stbt run` to be used if you want to use the library.  This would aid integration with other testing frameworks such as py.test.

By well-behaved I mean:

1. Has no global state
2. Can be cleanly initialised and completely deinitialised
3. Configuration can be injected rather than is loaded from disk implicitly

TODO (probably incomplete):

- [x] Create a `DeviceUnderTest` class refactoring global variables into it #285 
- [ ] Refactor `Display` to separate the sink-pipeline into a seperate class, allowing the sink-pipeline and the `DeviceUnderTest` objects to have independent lifetimes
- [ ] Inject all configuration into the `DeviceUnderTest` class, removing any calls to `get_config` outside of a factory function
- [ ] Remove `stbt.init_run` and `stbt.teardown_run`.
- [ ] Create a `contextmanager` or similar to take care of most of the concerns of `stbt run` and refactor `stbt run` to use it.

allanlewis, 2015-08-12:
Thanks @wmanley  - this will be useful for tracking progress towards librarification (is that a word?) of `stb-tester` :+1:

drothlis, 2016-07-11:
@martynjarvis @LewisHaley @jcutbush @johnteslade @allanlewis have you guys done any work on the above?

LewisHaley, 2016-07-11:
None at all.

wmanley, 2020-04-03:
I'm not planning on doing any more work on this for the time being, closing.

closed 2020-04-03 by wmanley

drothlis, 2020-04-03:
It's pretty much done already, at least for the use case of being able to `import stbt` for linting, IDEs, etc.