stb-tester/stb-tester #310: Bad error message when matching non-BGR images

LewisHaley, 2015-10-12:
```python
def _match(image, template, match_parameters, template_name):
    if any(image.shape[x] < template.shape[x] for x in (0, 1)):
        raise ValueError("Source image must be larger than template image")
    if any(template.shape[x] < 1 for x in (0, 1)):
        raise ValueError("Template image must contain some data")
    if template.shape[2] != 3:
        raise ValueError("Template image must be 3 channel BGR")
```

The final `if` causes an `IndexError` with single-channel images because the image `shape` is `(y, x)` not the assumed `(y, x, c)`. The result is that the purposefully placed error message about the required image types does not get raised.

wmanley, 2015-10-12:
I guess that should read `if len(template.shape) != 3 or template.shape[2] != 3:`.  Do you have a test case?  How did you come across this issue?

LewisHaley, 2015-10-12:
> Do you have a test case? How did you come across this issue?

I don't have a test case - I just was trying to get a fiddly bit of transparent navigation to work (which involved pre-processing frame and template) which resulted in passing in a greyscale (single channel) template into `stbt.match`. I think you could synthesise it with something like this:
```python
im = numpy.zeros(shape=(10, 10), dtype=numpy.uint8)
try:
    stbt.match(im)
except ValueError as e:
    assert "Template image must be 3 channel BGR" in str(e)
```

drothlis, 2015-10-15:
Thanks for the bugreport @LewisHaley. I've raised #315 to fix this.

commit 6cd9b727 (2015-10-15): stbt.match: Fix error message for greyscale images

This was raising "IndexError: tuple index out of range" instead of the
desired "ValueError: Template image must be 3 channel BGR".

Fixes #310.
PR #315 (2015-10-15): stbt.match: Fix error message for greyscale images
This was raising "IndexError: tuple index out of range" instead of the
desired "ValueError: Template image must be 3 channel BGR".

Fixes #310.
closed 2015-10-15 by drothlis