stb-tester/stb-tester #335: virtual-stb self-tests fail on Fedora

drothlis, 2016-02-26:
Reported by @LewisHaley on #334:

> I wasn't able to get `make check` to pass fully because of failures with the virtual-stb tests: 
> > /usr/libexec/Xorg.wrap: Only console users are allowed to run the X server

I don't have a Fedora host to test on right now, but some Googling found a similar issue[[1]] reported against xpra. Perhaps we could use their same workaround in this patch[[2]] -- that is, running `/usr/libexec/Xorg` instead of `/usr/bin/Xorg`.

[1]: https://www.xpra.org/trac/ticket/872
[2]: https://www.xpra.org/trac/attachment/ticket/872/xpra-0.14.22-fedora22-xorg.patch

LewisHaley, 2016-02-26:
From https://github.com/stb-tester/stb-tester/pull/334#issuecomment-189339337
> do you have xorg-x11-drv-dummy installed?

Yes, version 0.3.6-24.fc23, which is the latest version available on fc23.

Setting
```
▶ cat /etc/X11/Xwrapper.config
allowed_users = anybody
needs_root_rights = no
```
as suggested here https://bugs.archlinux.org/task/41391.
gets me further, but the tests fail because I don't have the `chromium-browser` executable (which seems to only be listed as a dependency in `.travis.yml`). `chromium-browser` is not available as-standard for Fedora.

drothlis, 2016-06-01:
In #360 I made the tests skip (instead of fail) if `chromium-browser` isn't installed.

@LewisHaley to run `stbt virtual-stb` from the console do you need your `Xwrapper.config` workaround, or was that only needed for the self-tests?

LewisHaley, 2016-06-01:
Running from stb-tester@362766c5c5ecadf13cac200645d1c763b3b6d84c, it still fails without the `Xwrapper.config` change:
```
▶ ./stbt_virtual_stb.py run echo lewis
/usr/libexec/Xorg.wrap: Only console users are allowed to run the X server
Traceback (most recent call last):
  File "./stbt_virtual_stb.py", line 147, in <module>
    sys.exit(main(sys.argv))
  File "./stbt_virtual_stb.py", line 116, in main
    x_keymap=args.x_keymap) as (child, config):
  File "/usr/lib64/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "./stbt_virtual_stb.py", line 51, in virtual_stb
    with x_server(1280, 720, verbose=verbose) as display:
  File "/usr/lib64/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "/home/lewis.haley/test-dev/uitests/stb-tester/_stbt/x11.py", line 88, in x_server
    % xorg.returncode)
_stbt.x11.XFailedToStartError: Failed to start X: Exited with status 1
```

drothlis, 2016-06-01:
Thanks @LewisHaley. I've updated the title of the issue to reflect that it also affects general usage.

wmanley, 2020-04-03:
This issue has been dormant for 4 years now, closing.  Please re-open if you still care.

closed 2020-04-03 by wmanley