stb-tester/stb-tester #365: Control device using CEC

Pooze, 2016-05-26:
I think it would be a great feature to have the possibility to control the connected device using the CEC feature in HDMI. It might at first look like a weird feature as most of the capture devices lacks CEC, however with an USB - CEC adapter ([for example](https://www.pulse-eight.com/p/104/usb-hdmi-cec-adapter#)) it would be possible to reach devices that cannot be controlled through IR nor IP.

In my case I'm facing a situation where I would like to use STB-Tester on Playstation 4, a device without IR nor IP controllability.

TODO (need to be broken down):

- [ ] Implement libCEC support within _stbt control_

wmanley, 2016-05-26:
That sounds like it could be really useful, particularly for the PS4 and other devices that don't have traditional remote controls like the Chromecast.

There are some licencing issues as libcec is GPLv2, whereas stb-tester is LGPLv2+.  As we would have to link against libcec we should be careful to ensure that the dependency is optional.  Perhaps it could be a separate Debian package so users who are happy with GPL test-packs can install the additional package.  I can help out with this bit.  I'll write some scaffolding that you can fill in.

I'm not sure what the best testing strategy is in this instance.  Perhaps mocking out the libcec API?  Usually I'm not a fan of mocking because it doesn't actually check that you've got the integration right, but it's better than no tests at all.  Unless you can think of a more representative test?

We look forward to your patches.

wmanley, 2016-05-26:
I've created the packaging scaffolding to help you on your way. It solves the boring build bits so you can get on with the functionality proper.  See the hdmi-cec-control branch of stb-tester:

https://github.com/stb-tester/stb-tester/commits/hdmi-cec-control

So the next step is to fill in the `HdmiCecControl` class from c87c71887326af1bbad10c1372b13bdb6ac8bbce.

Let us know by leaving a comment here if you need any more help.

alexeykorobov, 2016-05-30:
Hi there, I'm experimenting with the PS4 control as well. Have you managed to turn on the PS4 using the that USB-to-CEC device? I tried the CEC control with several TVs but haven't succeeded

danielandersson, 2016-06-01:
I just received the device and will start experimenting with it in the upcoming days.
Thanks a lot @wmanley for the scaffold and checklist-update.
License issues isn't really my cup of tea so I'm very happy you sorted it out. For the testing I agree with you, abstracting away the libCEC API is the best way. They have a great example client written in Python I expect could be used as reference when and then mocking its dependencies.

@alexeykorobov: I would prefer a solution with CEC as it could be used on other devices as well. However, it that, for some reason, won't work I'm glad you brought that idea up and it is definitely an interesting fix for PS4.

alexeykorobov, 2016-06-22:
Quick update on controlling the PS4 with HAMA's Speedshot keyboard/mouse converter and FLIRC: It doesn't work as I expected. 

Somehow, the FLIRC dongle doesn't get fully recognized by the Speedshot and the controls become inconsistent and unreliable. No problems when using the Speedshot with a standard wired keyboard though.

@danielandersson: how is your progress on the USB-CEC control with the new stbt branch?

danielandersson, 2016-06-27:
@alexeykorobov it is working. Some work has to be done before merged, see #378

drothlis, 2016-08-22:
Hi @danielandersson -- how are you getting on with using CEC as a control mechanism? Have you found it reliable enough for automated testing?

danielandersson, 2016-09-05:
Hi @drothlis, I have managed to get it up and running and in my experiments with Apple TV. However, I haven't had time to refine nor reviewed the code and unfortunately I'm not sure I'll in the near future. I think the CEC feature is of great benefit to the project and hope I'll find time to rework the code before making another merge request. 
Until then, anyone should feel free to continue my work with the CEC implementation.

danielandersson, 2016-10-21:
Closing, see issue #393.

closed 2016-10-21 by danielandersson