stb-tester/stb-tester #372: stbt batch: webm video capture doesn't adapt to video source changes (e.g. resolution)

alexeykorobov, 2016-06-08:
Looks like stbt's mechanism to capture webm videos relies solely on the parameters of the first frame of the video stream. This results in messed up video files when the format of the source changes dynamically, e.g. only the upper left corner of the screen is being captured.

Discovered with "source_pipeline = decklinkvideosrc mode=auto" that starts with an SD format but then immediately adapts to the actual format of the source, e.g. 1080p50. stbt run itself and stbt screenshot are not affected, as I can successfully execute stbt.match() on any part of the frame and take screenshot with the correct resolution. 

My environment: Ubuntu 16.04 x64, GStreamer 1.8.1, stb-tester v25

alexeykorobov, 2016-06-22:
I found a simple workaround by adding the following code to the source pipeline:
`videoscale ! video/x-raw,width=1280,height=720,pixel-aspect-ratio=1/1`

My pipeline config looks like this now:
`source_pipeline = decklinkvideosrc mode=auto ! videoscale ! video/x-raw,width=1280,height=720,pixel-aspect-ratio=1/1`

drothlis, 2017-07-24:
We recommend fixing the video-capture parameters, as per @alexeykorobov's second comment. Thanks for posting it!

closed 2017-07-24 by drothlis