stb-tester/stb-tester #381: Unknown display requirement

danielandersson, 2016-06-22:
Hey, recently  I have encountered a weird dependency.

I have the following test case:

```
def test_that_find_text():
    stbt.press('14:44:04')
    m = stbt.match_text("Text")
    assert m.match
```
When executing that locally on a machine it works fine. No problem. Similar to _stbt tv_ the screen pops up and then closes immediately.
However, when testing the exactly same test remotely over SSH the same test fails with the following error log:
```
FAIL: test.py::test_that_find_text: UITestError: gst-resource-error-quark: Could not initialise Xv output (10): Could not initialise Xv output
xvimagesink.c(1760): gst_xv_image_sink_open (): /GstPipeline:pipeline1/GstXvImageSink:xvimagesink0:
Could not open display (null)

Traceback (most recent call last):
  File "/usr/local/libexec/stbt/stbt-run", line 108, in <module>
    function()
  File "/home/user/stb/test.py", line 34, in test_that_find_text
    m = stbt.match_text("Text")
  File "/usr/local/libexec/stbt/stbt/__init__.py", line 412, in match_text
    text, frame, region, mode, lang, tesseract_config)
  File "/usr/local/libexec/stbt/_stbt/core.py", line 907, in match_text
    frame = self.get_frame()
  File "/usr/local/libexec/stbt/_stbt/core.py", line 949, in get_frame
    self._display.get_sample(), readonly=True) as frame:
  File "/usr/local/libexec/stbt/_stbt/core.py", line 1488, in get_sample
    raise UITestError(str(gst_sample))
UITestError: gst-resource-error-quark: Could not initialise Xv output (10): Could not initialise Xv output
xvimagesink.c(1760): gst_xv_image_sink_open (): /GstPipeline:pipeline1/GstXvImageSink:xvimagesink0:
Could not open display (null)
```

It seems to me that _match_text_ depends on having a window but I do not understand the reason to it since a test case as follows works, both over SSH and locally, without:

```
def test_experiment():
    match_param = stbt.MatchParameters()
    stbt.press('14:44:04')
    result = stbt.match('image-small.jpg', match_parameters=match_param)
    assert result
```

drothlis, 2016-06-22:
It sounds like you need to configure sink-pipeline to "fakesink sync=false" (either with stbt run's [--sink-pipeline] or in [stbt.conf]). More about sink-pipeline in the [GStreamer primer] on the wiki.

That you saw it with `match_text` but not with `match` is probably a race condition. I don't think there's any difference in the way `match` and `match_text` grab video.

[--sink-pipeline]: https://github.com/stb-tester/stb-tester/blob/master/docs/stbt.1.rst#global-options
[stbt.conf]: https://github.com/stb-tester/stb-tester/blob/master/docs/stbt.1.rst#configuration
[GStreamer primer]: https://github.com/stb-tester/stb-tester/wiki/Getting-started-with-stb-tester#appendix-gstreamer-primer

danielandersson, 2016-06-22:
Hmm, when setting sink-pipeline to "fakesink sync=false" and doing a _cv2.imwrite("screenshot.png", stbt.get_frame())_ the screenshot.png file is all black even tho, with the same settings, when using _stbt tv_ the screen is visible and everything looks to be in order.

danielandersson, 2016-06-22:
*missclosed it

I found a dirty solution for it.
```
def test_screenshot():
    import os
    while stbt.is_screen_black():
        print "Screen is black"
    cv2.imwrite("screenshot.png", stbt.get_frame())
    print os.path.getsize('screenshot.png')
```

drothlis, 2016-06-22:
Sounds to me like a fault with your video-capture hardware -- it seems to deliver some black frames before delivering the real frames.

danielandersson, 2016-06-27:
I think this is the same issue Alexey mention [here](https://groups.google.com/d/msg/stb-tester/Q6MJ-4m81M4/eiCWaSchAwAJ) as bullet number 4.

`gst-launch-1.0 decklinkvideosrc mode=auto ! videoconvert ! pngenc snapshot=true ! filesink location="Pictures/YourFileName.png"`

Gives the same black frames.

alexeykorobov, 2016-06-27:
@danielandersson: for taking screenshots, I found the "stbt screenshot" to be way easier than performing acrobatics with the GStreamer pipeline. It has quite an advanced pipeline config to skip the first frames, I assume it's due to an intelligent use of several queues.

danielandersson, 2016-06-27:
That is actually true, very interesting. Maybe similar pipeline can be used to overcome the black frames when running _stbt run_

drothlis, 2016-06-28:
I'm unlikely to accept any code in stbt-run or _stbt.core.Display to add
special case workarounds for buggy hardware. We've added workarounds
before (e.g. the source-pipeline restarting logic) and it has made the
code very complicated, difficult to work with & difficult to test.

Our default position is that any workarounds need to be done in your
source pipeline. For example you might find an existing GStreamer filter
element that can be configured to drop the first few frames that it
sees. If it doesn't exist, you could write one, or you could add the
workaround to the decklink source element.

danielandersson, 2016-06-29:
@drothlis I understand that. Maybe a quick clean-up with those legacy workarounds should be in place or another plugin-based approach might be beneficial.

The issue, from what I have found is that the first frame is a 720x486px all black and a working workaround for this is to call `get_frame()` one additional time.
The exact reason for this black frame is still unknown. Most likely caused by the GStreamer plugin _decklinkvideosrc_.

closed 2016-06-29 by danielandersson