stb-tester/stb-tester #385: raise NoVideo("No video") _stbt.core.NoVideo: No video error

ahmedusman2, 2016-07-18:
Hi,

Currently i am using stbt on ubantu Release:    15.04 with kernal info 3.19.0-49-generic with black magic card. i was able to set end to end test environment. but after couple of script records "stbt record" command has stopped working and shows following error:

  root@Linux-1:~# stbt record
Traceback (most recent call last):
File "/usr/lib/stbt/stbt-record", line 78, in <module>
sys.exit(main(sys.argv))
File "/usr/lib/stbt/stbt-record", line 46, in main
record(args.control_recorder, script)
File "/usr/lib/stbt/stbt-record", line 52, in record
stbt.get_frame()  # Fail early if no video
File "/usr/lib/stbt/stbt/__init__.py", line 406, in get_frame
return _dut.get_frame()
File "/usr/lib/stbt/_stbt/core.py", line 900, in get_frame
self._display.get_sample(), readonly=True) as frame:
File "/usr/lib/stbt/_stbt/core.py", line 1441, in get_sample
raise NoVideo("No video")
_stbt.core.NoVideo: No video

hardware.conf

REMOTE="New_AVDevice"
REMOTE_MODULES=""
REMOTE_DRIVER=""
REMOTE_DEVICE="/dev/lirc0"
REMOTE_SOCKET=""
REMOTE_LIRCD_CONF=""
REMOTE_LIRCD_ARGS=""
TRANSMITTER="None"
TRANSMITTER_MODULES=""
TRANSMITTER_DRIVER=""
TRANSMITTER_DEVICE=""
TRANSMITTER_SOCKET=""
TRANSMITTER_LIRCD_CONF=""
TRANSMITTER_LIRCD_ARGS=""
START_LIRCD="true"
START_LIRCMD=""
LOAD_MODULES=""
LIRCMD_CONF=""
FORCE_NONINTERACTIVE_RECONFIGURATION=""

stbt.conf
[global]
source_pipeline = decklinksrc connection=hdmi mode=720p50 device-number=0 ! videoconvert
sink_pipeline=xvimagesink sync=false
transformation_pipeline = identity
control=lirc::New_AVDevice
verbose=0
power_outlet=none
v4l2_ctls=

# Handle loss of video (but without end-of-stream event) from the video capture
# device. Set to "True" if you're using the Hauppauge HD PVR.
restart_source = False

# stbt camera settings that have to be in global as Python's configparse
# doesn't allow substitutions between sections.
geometriccorrection_params =
contraststretch_params =

[camera]
# Format of the calibration videos that are generated by "stbt camera
# calibrate" to be played on your TV as part of the camera calibration process.
# Valid values are "ts" (for MPEG-TS) and "mp4". This is configurable because
# my Panasonic TV doesn't like MPEG-TS and my Samsung TV doesn't like MP4.
video_format = mp4

[match]
match_method=sqdiff-normed
match_threshold=0.80
confirm_method=absdiff
erode_passes=1
confirm_threshold=0.16

# Downsample the video frame and the template image before matching, as a
# performance optimisation. Once found, the match is always confirmed against
# the full-sized images, so this should never affect the outcome of a match,
# only its speed. Set to `1` to disable this optimisation.
pyramid_levels = 3

[press]
interpress_delay_secs = 0

[press_until_match]
interval_secs = 3
max_presses = 10

[motion]
noise_threshold=0.84
consecutive_frames=10/20

[is_screen_black]
threshold = 10

[run]
save_video =video.webm

[record]
output_file=test.py
control_recorder=lirc::New_AVDevice

drothlis, 2016-07-18:
On Mon, Jul 18, 2016, at 08:05 AM, ahmedusman2 wrote:
> Hi,
> Currently i am using stbt on ubantu Release:    15.04 with kernal
> info 3.19.0-49-generic with black magic card. [...]
> _stbt.core.NoVideo: No video
 
Looks like your video-capture hardware crashed / froze. It's a common
problem with Blackmagic hardware. Or maybe you have some other process
still running that is reading from the device -- again your hardware
doesn't support multiple simultaneous readers.
 
Kind regards,
 
David Röthlisberger
Stb-tester.com Ltd.
 
--
Stb-tester.com Ltd is a company registered in England and Wales.
Registered number: 08800454. Registered office: 33b Aliwal Road,
Battersea, London, SW11 1RB.

ahmedusman2, 2016-07-18:
thanks david for your reply. can you please suggest any way to recover this crashing. Or can you suggest any other such card that works fine with STBT-Tester.

Thanks

drothlis, 2016-07-18:
Yes, we recommend the [stb-tester ONE](https://stb-tester.com/stb-tester-one).

closed 2016-07-18 by drothlis