stb-tester/stb-tester #403: Warning Dropped frames  and test case fail

jobitmathew, 2016-12-28:
I am running some test cases to check the image comparison after record  and playing back video.But some time issue getting as "Dropped frames " and test case fail.Same test case may pass next time rerun or after reboot.I am attaching the screen shot of issue.Anyone faced this issue?and any solution is there.
[image]
I am using Blackmagic intensity pro card.
root@huawei-ThinkCentre-M91P:~# lspci | grep Blackmagic
01:00.0 Multimedia video controller: Blackmagic Design Intensity Pro

and settings are
[global]
source_pipeline=decklinksrc connection=hdmi mode=720p50  ! videoconvert
sink_pipeline=xvimagesink sync=false
transformation_pipeline = identity
control=lirc:8700:huawei
verbose=0
power_outlet= IPS
v4l2_ctls=

[match]
match_method=sqdiff-normed
match_threshold=0.80
confirm_method=absdiff
erode_passes=1
#confirm_threshold=0.16
confirm_threshold=0.20
pyramid_levels = 3

[press]
interpress_delay_secs = 2

[press_until_match]
interval_secs = 3
max_presses = 10

[motion]
noise_threshold=0.84
consecutive_frames=10/20

[is_screen_black]
#threshold = 10
threshold = 30

danielandersson, 2017-01-05:
@jobitmathew what version of GStreamer are you using?

jobitmathew, 2017-01-05:
please find the version details 
[image]

drothlis, 2017-01-24:
Hi @jobitmathew, probably more people watch the [mailing list](http://groups.google.com/group/stb-tester) than the github issues.

I think "Dropped frames" is being reported by the Blackmagic driver / decklinksrc. It means your computer isn't processing the frames fast enough so it's skipping some frames (not delivering them to stb-tester). That shouldn't affect your test -- stb-tester will just process the next frame that is available.

The test failure is probably unrelated to the "dropped frames" warnings. You didn't show us the output of the test when it failed. `stbt run` should also save a screenshot when the test failed.

closed 2017-06-26 by drothlis