stb-tester/stb-tester #448: Video Capture Device

ManinderBoya, 2017-07-24:
Hi There,

Thanks for the information on Wiki page.

I have query regarding video capturing device to be user with STB tester.
We have "Hauppauge Colossus 2 PCI Express Internal 1080p HD-PVR". is this the same as the device you mentioned (HD PVR model 1212) on wiki?

Can we use above mentioned PCI based device with STB Tester? 
Do we need to have any extra configurations to be done?
With the above PCI device, will all the API's works as expected like capture frame, video , OCR on live etc ?

Please help in answering above queries.

Thanks a lot in advance

drothlis, 2017-07-24:
Hi @ManinderBoya 

> We have "Hauppauge Colossus 2 PCI Express Internal 1080p HD-PVR". is this the same as the device you mentioned (HD PVR model 1212) on wiki?

No, it isn't the same device.

> Can we use above mentioned PCI based device with STB Tester?

I don't know, I have never used that device. You could try asking on the mailing list, maybe someone in the community there has used your device.

If you want a tested, guaranteed, and **reliable** video-capture hardware that works without any configuration needed, I recommend our own stb-tester hardware [here](https://stb-tester.com/solutions).

closed 2017-07-24 by drothlis

ManinderBoya, 2017-07-24:
@drothlis 

Can i able to buy only STB Tester HW video capture excluding all other features? by paying one time ?
If yes, how much would that cost?

drothlis, 2017-07-24:
No, I'm afraid that the product offerings on our website are all we have. Contact sales@stb-tester.com if you want to discuss further.

ManinderBoya, 2017-07-24:
Thanks @drothlis