stb-tester/stb-tester #477: Useful new lint ideas

wmanley, 2017-12-22:
Some ideas for lints inspired by code seen in the wild:

## assert-true-has-no-effect

```
assert True
```

`assert True` statement has no effect.  Remove this line or replace it with `pass`

```
assert True, "Things are fine"
```

`assert True` statement has no effect.  Perhaps you meant `logging.info("Things are fine")`

## assert-unnecessary-if

```
if stbt.match('abcd.png'):
    ...
else:
    assert False, "Didn't find abcd menu"
```

Unnecessary `if` statement. Consider rewriting with `assert`: `assert stbt.match('abcd.png'), "Didn't find abcd menu"`

But this code is fine:

```
if stbt.match('abcd.png'):
    ...
elif stbt.match('efgh.png'):
    ...
else:
    assert False, "Couldn't find menu"
```

## return-unnecessary-if

This one is possibly more contentious:

```
if stbt.match('abcd.png'):
    return True
else:
    return False
```

This is preferred:

```
return bool(stbt.match('abcd.png'))
```

drothlis, 2018-05-16:
## slow-is-visible

Does `is_visible` use `stbt.match_text` or `stbt.ocr` as its first operation? You can often discard 90% of false-positives by doing a fast match operation first, for example:

    return (stbt.match("image-that-is-present-in-this-and-some-other-screens.png") and 
            stbt.match_text("text that is only present in this screen"))

wmanley, 2018-09-20:
## Public FrameObject property returns `MatchResult` object

...or possibly another object that has a `.frame` member?

```
class Example(stbt.FrameObject):
    @property
    def is_good(self):
        return stbt.match("good.png")
```

Either explicitly make it a `bool()`

```
class Example(stbt.FrameObject):
    @property
    def is_good(self):
        return bool(stbt.match("good.png"))
```

or make the property private:

```
class Example(stbt.FrameObject):
    @property
    def _is_good(self):
        return stbt.match("good.png")
```