stb-tester/stb-tester #552: No xenial version in ppa:stb-tester/stb-tester

zuzav, 2019-01-10:
I have realized today that it is not possible to install stb-tester for Xenial Ubuntu from ppa . I don't know if it is intentional?

drothlis, 2019-01-10:
The v29 packages are currently broken. Until someone fixes them you'll need to download the v28 deb and install it manually, or build & install from source. See See <https://groups.google.com/d/msg/stb-tester/AbHQV26Lh3k/xexImGd0EQAJ>.

msiden, 2019-02-11:
If anyone would care to explain how to build and install from source that would be much appreciated. I've been trying all morning but after install stbt won't run.

msiden, 2019-02-11:
@drothlis The debs for v28 are missing and building from source doesn't seem to work either (unless I'm doing it wrong).

drothlis, 2019-02-12:
To build & install from source:

* Check out the version you want (for example `git checkout v28`).
* `make`
* `sudo make install`

msiden, 2019-02-12:
Thank you David! Yes, I tried that a number of times with no luck. I managed to figure out how to create debs with ubuntu-shell.sh though so problem solved (and I will now keep my own local copy of the debs to avoid having to go through this again) :)

drothlis, 2019-02-22:
v30 debs are now available in the ppa for Ubuntu 16.04 (xenial) and 18.04 (bionic).

closed 2019-02-22 by drothlis