stb-tester/stb-tester #556: Support for RedRat X

msiden, 2019-01-23:
I would like to add support for the Red Rat X IR blaster to the official release of stbt. I've been using this device with stbt for about a year now and have found that it works just fine when "emulating" an IRNetBox MK2 so this is my proposed solution. It would NOT work with USB in other words. I have been in touch with the people at RedRat who have confirmed that this solution should work since the RedRat X is basically a scaled down IRnetbox 4.

Looking at the code in irnetbox.py it's a bit unclear to me whether IRNetBox MK3 and 4 are supported. From what I can tell a different command is send to these devices in irsend_raw. It would help if I could get a better understanding of this.

msiden, 2019-01-24:
@drothlis I have made some small updates to irnetbox.py that integrates support for the RedRat X. I have tested it and it works fine. Seems I need permission to commit my code and make a pull request though.

msiden, 2019-01-24:
Done. See pull request #557.

msiden, 2019-01-24:
Speaking to Chris at RedRat this implementation _should_ also work with IRNetBox 2, 3, and 4. I can't confirm this though since I don't have access to these devices and can't test them.

drothlis, 2019-05-09:
Implemented in #557.

closed 2019-05-09 by drothlis