stb-tester/stb-tester #577: How to import "stbt" and its functions into PyCharm?

mattissf, 2019-02-28:
Hi

Maybe I am missing something obvious, but I can not for the life of me figure out how to import "stbt" properly into my PyCharm test-pack project.

[image]

How should I do this?

msiden, 2019-02-28:
A simple "import stbt" should definitely work. If it doesn't I figure there must be something wrong with your installation or your environment. Have you checked your Pythonpath?

mattissf, 2019-02-28:
I think my problem is that I use a project based off of "stb-tester-test-pack" that does not include "stbt" as a library. I don't have STB tester installed locally, since I use pipenv, and I don't think this library is available on pypi.org?

I ended up unpacking release 29 of stb-tester and just adding it as a source folder in my PyCharm project. That seemed to work.

mattissf, 2019-02-28:
I will close this ticket here, thanks for the help!

closed 2019-02-28 by mattissf