stb-tester/stb-tester #590: Color recognition

msiden, 2019-05-13:
I always thought it would be great if stbt could identify what color is displayed in a particular region of the screen. I added this functionality to my own stbt-based framework a while back and it's been very useful and a great alternative in scenarios where image recognition can be unreliable in my experience. 

Use cases:
* Verify that the background or a specific region changes color when you enter certain menus or screens
* Verify that menu text changes color when selected

This function works as follow:
* Grab a frame from the device under test
* Extract the specified region of the frame. This can be as small as one pixel. It's recommended to keep regions as small as possible for maximum accuracy and speed
* Check the BGR values for each pixel and divide by the total number of pixels. In other words, it "blends" all pixels and returns the average.
* Return result as a tuple (B, G, R)

This would result in two new methods in stbt
* stbt.get_color - Takes a region argument and returns a tuple as described above
* stbt.is_color - Takes two arguments, region and color_range which is a tuple containing the lowest and highest allowed values for the expected color ((0, 0, 0), (70, 70, 70)). Returns a boolean.

This is only a few lines of code that I would be happy to share with the community if deemed interesting.

drothlis, 2019-05-17:
That sounds useful. One specific use-case is for UIs where the selection is shown by changing the colour of the button or menu entry (instead of drawing a box around it).

We've implemented similar things in our test-packs -- I think we were looking for the dominant colour (the most frequent colour in the region) rather than the average. Or sometimes the brightest colour. It depends on what your UI looks like and what you're trying to accomplish.

A similar thing we've done is to detect a spinner like this:

[image]

The 4 blobs spin around so you can't use `stbt.match` to find it. The background can show live TV, not just black. What we did was check that all 4 colours were present in that region, with at least some number of pixels for each particular colour. This worked very reliably (and fast).

If you share some examples of your use-cases, we might be able to find a solution that's general. In any case, please do share your code -- even if it isn't general enough to merge, it can serve as an example for other people to use in their test-packs.

msiden, 2019-05-17:
I opened a pull request https://github.com/stb-tester/stb-tester/pull/592
I think that should give you a good idea of what I had in mind.