stb-tester/stb-tester #595: Python 3 support

drothlis, 2019-05-22:
Allow running Python 2 or 3 test scripts from the same stb-tester codebase.

- [x] Use Python 3 future imports (print function, unicode literals, etc) and backported builtins everywhere. Get pylint passing on both 2 & 3. #583 
- [ ] Get unit tests passing on Python 3. #594 
- [ ] Get integration tests passing on Python 3 (`make check-integrationtests` -- this tests integration with GStreamer, via the `stbt run` command-line interface; but it doesn't test against real video-capture or lirc hardware).
- [ ] Test `stbt.press()` and `stbt control` with LIRC hardware.
- [ ] Test `stbt.press()` and `stbt control` via stbt-control-relay with LIRC hardware.
- [ ] Python 3: `print` output is duplicated (add test case & fix it).

Post-merge todo items from #583:

- [x] Revisit pylint:disable=stop-iteration-return ([comment](https://github.com/stb-tester/stb-tester/pull/583#discussion_r283794786), [comment](https://github.com/stb-tester/stb-tester/pull/583#discussion_r283798358)) #593 #594.
- [ ] Revisit raise AdbError as bytes ([comment](https://github.com/stb-tester/stb-tester/pull/583#discussion_r283804013))
- [ ] Revisit android._adb return type ([comment](https://github.com/stb-tester/stb-tester/pull/583#discussion_r283805392))
- [x] Check if `struct.pack` first argument has to be bytes ([comment](https://github.com/stb-tester/stb-tester/pull/583#discussion_r283822226))
- [ ] Avoid decoding + re-encoding byte-string tesseract_user_words & patterns ([comment](https://github.com/stb-tester/stb-tester/pull/583#discussion_r283823127))
- [ ] Should subprocess arguments be bytes? ([comment](https://github.com/stb-tester/stb-tester/pull/583#discussion_r283827316))
- [x] Change regexes to bytes in stbt_lint.py ([comment](https://github.com/stb-tester/stb-tester/pull/583#discussion_r283828269))
- [x] Do to_bytes later in videotestsrccontrol.press ([comment](https://github.com/stb-tester/stb-tester/pull/583#discussion_r283835162))

drothlis, 2019-05-28:
Hmmm [six] has more recent releases than python-future. I wonder why I thought the latter was the replacement for six? Oh well it's done now.

[six]: https://pypi.org/project/six/

drothlis, 2019-05-30:
@wmanley how important do you think the remaining todo items (above) are?

wmanley, 2019-05-30:
I don't think any of them are particularly important.  But equally they don't seem like they're big tasks.

drothlis, 2019-09-18:
This has been well tested with several customer test-packs using v31 Beta.

closed 2019-09-18 by drothlis