stb-tester/stb-tester #608: Cannot define __init__() method at the beginning of a class

aleksanderczeszyk-intive, 2019-07-30:
I would like to create a def __init__(self) method at the beginnig of a FrameObject inheriting class.

```
class AppleTvHomeScreen(stbt.FrameObject):

    def __init__(self):
        self.appleTvHomeIcon = "images/appletv/apple-tv-home-icon.png"
        self.appleTvHomeYoutubeIcon = "images/appletv/apps/youtube/apple-home-youtube.png"

    @property
    def is_visible(self):
        return stbt.match(self.appleTvHomeIcon,
                          frame=self._frame, region=self.selection_region)

    @property
    def selection_region(self):
        return stbt.Region(x=36, y=483, width=249, height=152)  # Fixed focus

    def go_back_to_home(self):
        for _ in range(8):
            stbt.press("KEY_BACK")
            isHomeVisible = stbt.wait_until(AppleTvHomeScreen, timeout_secs=2)
        assert isHomeVisible
```
It is just a work in progres code to make it work. When running go_back_to_home() method i get an error:
**AttributeError: 'AppleTvHomeScreen' object has no attribute '_FrameObject__local'**
The error does not occur when the init method is removed and the paths are put into the properties (they are not class attributes). I don't have any clue why would that work like this. Can anyone help?

allanlewis, 2019-07-30:
You almost certainly need to call the parent classes `__init__` from your `__init__`:
```py
super(AppleTvHomeScreen, self).__init__(self)
```
You probably want to do that at the top of your custom constructor.

drothlis, 2019-07-30:
@allawlewis is correct, but you also need to accept a `frame` parameter and pass it through to the base class, like this:

```
class AppleTvHomeScreen(stbt.FrameObject):
    def __init__(self, frame=None):
        super(AppleTvHomeScreen, self).__init__(self, frame)
        ...now your own code here...
```

In general, a subclass's `__init__` always must call the base class `__init__` first. Sometimes the base class's `__init__` doesn't do anything so you get away with it. But in this case, `stbt.FrameObject` provides a lot of special behaviour (caching of properties, etc -- see the [API docs]).

[API docs]: https://stb-tester.com/manual/python-api#stbt.FrameObject

allanlewis, 2019-07-30:
Thanks for the correction, @drothlis - I forgot about the `frame` parameter.

aleksanderczeszyk-intive, 2019-07-30:
Thanks, it worked! Closing the issue

closed 2019-07-30 by aleksanderczeszyk-intive