stb-tester/stb-tester #648: OpenCV 4 Support

eugenetriguba, 2019-12-22:
I've been using the library with OpenCV 4, but it just required some small changes to ``cv2_compat.py`` to get things working. I could submit a pull request for it, just wanted to gauge if this is something that would be desired.

https://stackoverflow.com/questions/54734538/opencv-assertion-failed-215assertion-failed-npoints-0-depth-cv-32

The stack overflow post explains the issue, related to changes to what ``findContours`` returns.

drothlis, 2019-12-22:
> I could submit a pull request for it

Yes please.

demorph, 2020-01-13:
cv2_compat.py : line 18
contours = cv2.findContours(image=image, mode=mode, method=method)[0]
fixed it  =)

drothlis, 2020-02-12:
Fixed in #650.

closed 2020-02-12 by drothlis