stb-tester/stb-tester #657: stbt v31 always runs in Python 2

msiden, 2020-02-27:
I've recently upgraded to stbt v31 and have ported my test pack to Python 3 (after carefully reading this: https://stb-tester.com/blog/2019/09/20/porting-to-python3). 

I added these lines to ~/.config/stbt/stbt.conf
```
[test_pack]
stbt_version = 31
python_version = 3
```

Typing 
`
stbt config test_pack.python_version
`
returns "3"

However when I run my tests with stbt run it actually run in Python 2 and I get a SyntaxError:
FAIL: test1.py: SyntaxError: invalid syntax
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/_stbt/stbt_run.py", line 139, in sane_unicode_and_exception_handling
    yield
  File "/usr/lib/stbt/stbt_run.py", line 52, in main
    test_function.call()
  File "/usr/lib/python2.7/dist-packages/_stbt/stbt_run.py", line 117, in fn
    exec(code, test_globals)  # pylint:disable=exec-used

What am I missing?

drothlis, 2020-03-02:
Hi @msiden 

The instructions you followed are for our customers who are using the commercial Stb-tester Platform.

If you're running Stb-tester from the command line then you have to run it with python3 explicitly:

    python3 /path/to/stbt_run.py ...

You'll also have to install any Python 3 dependencies yourself, because the PPA package only installs the Python 2 dependencies.

closed 2020-03-02 by drothlis

msiden, 2020-03-05:
Hi @drothlis 
Ok, thanks for clarifying. It can often get quite confusing with these features to know what applies to the commercial platform and what applies to us open source users.

drothlis, 2020-03-05:
@msiden the https://stb-tester.com website and blog is 100% aimed at our customers.

For the open source codebase please refer to the git log, the release notes in the source repo (https://github.com/stb-tester/stb-tester/blob/master/docs/release-notes.md) and the mailing list.