stb-tester/stb-tester #673: libcec and Python 3

msiden, 2020-05-14:
I have got stbt up and running with Python 3. Everything seems to be working well except from when I use cec as control input. When I try to run a test I get this exception:
`AttributeError:` module 'cec' has no attribute 'libcec_configuration'

Apparently there is a bug in libcec 4.0.2 which has been fixed in 4.0.4: https://bugs.launchpad.net/ubuntu/+source/libcec/+bug/1805620

Could I solve this issue by upgrading to Ubuntu 20? I'm now on Ubuntu 18.

Any advice is welcome. I've spent many hours investigating this and I'm just out of ideas.

drothlis, 2020-05-15:
As far as I know, Ubuntu doesn't have Python3 bindings for libcec.

> Apparently there is a bug in libcec 4.0.2 which has been fixed in 4.0.4: https://bugs.launchpad.net/ubuntu/+source/libcec/+bug/1805620

Stb-tester already works around this bug (in fact it was my colleague @wmanley who reported it).

msiden, 2020-05-27:
I did some more digging into this issue and thought I'd share my findings.

As mentioned above stbt works fine with Python 3 under Ubuntu 18 as long as you don't need to use cec. (You have to install from source though).

Getting cec to work was trickier but I finally managed to work it out by upgrading to Ubuntu 20 and installing the latest version of libcec (4.0.4) which comes with Python 3 bindings. 

This however caused stbt to raise an exception in the control_gpl module (TypeError: in method 'libcec_configuration_strDeviceName_set', argument 2 of type 'char [13]'). From what I can gather this was caused by stbt trying to send a byte string to libcec. I did a small modification to the control_gpl module sending regular unicode strings instead and this solved the issue. I'd be happy to make a pull request with my modifications if you're interested. 

This solution means of course that stbt will only work with Python 3 and not support Python 2.

drothlis, 2020-05-27:
Great, thanks for sharing.

A pull request is welcome but it will need to work with both Python 2 and 3. You can use `to_native_str()` in `_stbt/utils.py`.

msiden, 2020-06-09:
Opened a pr: https://github.com/stb-tester/stb-tester/pull/677

commit 3ac10600 (2020-06-11): HdmiCecControl: Fix for Python 3

tested ok on Python 2.7 in Ubuntu 18 and Python 3.8 in Ubuntu 20

Fixes #673.
closed 2020-06-11 by drothlis