stb-tester/stb-tester #789: match on single channel images ignores template alpha

wmanley, 2023-04-17:
In `match` if frame is a single channel array (like from `color_diff`) then alpha in the template is ignored:

https://github.com/stb-tester/stb-tester/blob/71b907aa4a3e5208843cc63dc137d818818ce83c/_stbt/match.py#L354

So this behaves incorrectly:

```py
diff = color_diff(self._frame, "#ffffff")
stbt.match("template-with-alpha.png", frame=diff)
```

It should behave as if `diff` had three channels.

Workaround:

```py
diff = color_diff(self._frame, "#ffffff")
diff = load_image(diff, color_channels=3)
stbt.match("template-with-alpha.png", frame=diff)
```

drothlis, 2023-04-25:
How to tell if the reference image has an alpha channel?

Option 1: Just load the entire image with `cv2.IMREAD_UNCHANGED` and then convert to grayscale if it doesn't have an alpha channel.

Option 2: Look at the PNG headers.

Then, if the image does have an alpha channel, convert the frame from grayscale to BGR? That would be less optimised, but less risky, than changing `match` to handle reference images with 2 channels (grayscale + alpha).

Option 1 is probably fine (`load_image` is memoized). However out of curiosity I looked at the PNG spec:

## PNG spec

https://www.w3.org/TR/png/#5DataRep:

> ## 5\. Datastream structure
> 
> ### 5.1 Introduction
> 
> The PNG datastream consists of a PNG signature followed by a sequence of chunks.
> 
> ### 5.2 PNG signature
> 
> The first eight bytes of a PNG datastream always contain the following (decimal) values:
> 
>     137 80 78 71 13 10 26 10
> 
> which are (in hexadecimal):
> 
>     89 50 4E 47 0D 0A 1A 0A
> 
> This signature indicates that the remainder of the datastream contains a single PNG image, consisting of a series of chunks beginning with an [IHDR][4] chunk and ending with an [IEND][5] chunk.
> 
> ### 5.3 Chunk layout
> 
> Each chunk consists of three or four fields (see [Figure 10][7]). The meaning of the fields is described in [Table 4][8]. The chunk data field may be empty.
> 
> Figure 10 Chunk parts
> 
> <table id="table51" class="numbered simple"><caption>Table <bdi class="tableno">4</bdi> <span class="table-title">Chunk fields</span></caption><tbody><tr><th>Name</th><th>Description</th></tr><tr><td>Length</td><td>A <a href="https://www.w3.org/TR/png/#dfn-png-four-byte-unsigned-integer" class="internalDFN" data-link-type="dfn" id="ref-for-dfn-png-four-byte-unsigned-integer-1">PNG four-byte unsigned integer</a> giving the number of bytes in the chunk's data field. The length counts <strong>only</strong> the data field, <strong>not</strong> itself, the chunk type, or the <a href="https://www.w3.org/TR/png/#dfn-crc" class="internalDFN" data-link-type="dfn" id="ref-for-dfn-crc-1">CRC</a>. Zero is a valid length. Although encoders and decoders should treat the length as unsigned, its value shall not exceed 2<sup>31</sup>-1 bytes.</td></tr><tr><td>Chunk Type</td><td>A sequence of four bytes defining the chunk type. Each byte of a chunk type is restricted to the decimal values 65 to 90 and 97 to 122. These correspond to the uppercase and lowercase ISO 646 [<cite><a class="bibref" data-link-type="biblio" href="https://www.w3.org/TR/png/#bib-iso646" title="Information technology — ISO 7-bit coded character set for information interchange">ISO646</a></cite>] letters (<code>A</code>-<code>Z</code> and <code>a</code>-<code>z</code>) respectively for convenience in description and examination of PNG datastreams. Encoders and decoders shall treat the chunk types as fixed binary values, not character strings. For example, it would not be correct to represent the chunk type <a class="chunk" href="https://www.w3.org/TR/png/#11IDAT">IDAT</a> by the equivalents of those letters in the UCS 2 character set. Additional naming conventions for chunk types are discussed in <a href="https://www.w3.org/TR/png/#5Chunk-naming-conventions" class="sec-ref"><bdi class="secno">5.4 </bdi>Chunk naming conventions</a>.</td></tr><tr><td>Chunk Data</td><td>The data bytes appropriate to the chunk type, if any. This field can be of zero length.</td></tr><tr><td>CRC</td><td>A four-byte <a href="https://www.w3.org/TR/png/#dfn-crc" class="internalDFN" data-link-type="dfn" id="ref-for-dfn-crc-2">CRC</a> calculated on the preceding bytes in the chunk, including the chunk type field and chunk data fields, but <strong>not</strong> including the length field. The <a href="https://www.w3.org/TR/png/#dfn-crc" class="internalDFN" data-link-type="dfn" id="ref-for-dfn-crc-3">CRC</a> can be used to check for corruption of the data. The <a href="https://www.w3.org/TR/png/#dfn-crc" class="internalDFN" data-link-type="dfn" id="ref-for-dfn-crc-4">CRC</a> is always present, even for chunks containing no data. See <a href="https://www.w3.org/TR/png/#5CRC-algorithm" class="sec-ref"><bdi class="secno">5.5 </bdi>CRC algorithm</a>.</td></tr></tbody></table>
> 
> The chunk data length may be any number of bytes up to the maximum; therefore, implementors cannot assume that chunks are aligned on any boundaries larger than bytes.
> 
> [1]: https://www.w3.org/TR/png/#5DataRep
> [2]: https://www.w3.org/TR/png/#5Introduction
> [3]: https://www.w3.org/TR/png/#5PNG-file-signature
> [4]: https://www.w3.org/TR/png/#11IHDR
> [5]: https://www.w3.org/TR/png/#11IEND
> [6]: https://www.w3.org/TR/png/#5Chunk-layout
> [7]: https://www.w3.org/TR/png/#figure411 "Chunk parts"
> [8]: https://www.w3.org/TR/png/#table51 "Chunk fields"
> [9]: https://www.w3.org/TR/png/#dfn-png-four-byte-unsigned-integer
> [10]: https://www.w3.org/TR/png/#dfn-crc
> [11]: https://www.w3.org/TR/png/#bib-iso646 "Information technology — ISO 7-bit coded character set for information interchange"
> [12]: https://www.w3.org/TR/png/#11IDAT
> [13]: https://www.w3.org/TR/png/#5Chunk-naming-conventions
> [14]: https://www.w3.org/TR/png/#dfn-crc
> [15]: https://www.w3.org/TR/png/#dfn-crc
> [16]: https://www.w3.org/TR/png/#dfn-crc
> [17]: https://www.w3.org/TR/png/#5CRC-algorithm

https://www.w3.org/TR/png/#11IHDR:

> #### 11.2.1 IHDR Image header
> 
> The four-byte chunk type field contains the decimal values
> 
>     73 72 68 82
> 
[... cut ...]
> [1]: https://www.w3.org/TR/png/#6Colour-values
> [2]: https://www.w3.org/TR/png/#4Concepts.PNGImage
> [3]: https://www.w3.org/TR/png/#3truecolour
> [4]: https://www.w3.org/TR/png/#3truecolour

Note that there are other ways of specifying alpha: "Indexed color", where each pixel is an index into a table of colours (which can include alpha); or a separate [tRNS](https://www.w3.org/TR/png/#11tRNS) chunk which can specify that a single RGB value is to be treated as "fully transparent". We can probably ignore both of these.

drothlis, 2023-04-25:
I raised #792 with a testcase. I haven't implemented the actual fix yet.

closed 2023-04-25 by drothlis